Florin-Popescu/avarice-unofficial #2: Compatibility with Arduino Nano Every

maxgerhardt, 2021-02-12:
I'm unsure how to start avarice for my particular board.

It's a Arduino Nano Every with a ATMega4809 chip. It has a SAMD11 chip on board as a programmer. I can flash the firmware fine using avrdude with

>avrdude -v -p atmega4809 -C "C:\Users\Max\.platformio\packages\tool-avrdude-megaavr\avrdude.conf" -c jtag2updi -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x00:m -Ulock:w:0xc5:m -b 115200 -V -e -D -P "COM20" -U flash:w:.pio\build\nano_every\firmware.hex:i

So some firmware must be flashed on the SAMD11 that implements that JTAG2UPDI protocol The Arduino page says it's the [Arduino MuxTO](https://www.arduino.cc/en/Guide/NANOEvery#firmware-for-samd11d14a) bootloader. 

I can compile this program on linux just fine (just execute the last few lines in the build sh script), but when I start it with what I think is right I get

```
$ sudo ./avarice -j /dev/ttyACM0  -d --part atmega4808  :4242  
AVaRICE version 2.14svn20200906, Feb 12 2021 17:43:24

Defaulting JTAG bitrate to 250 kHz.

JTAG config starting.
Attempting synchronisation at bitrate 115200

command[S, 1]: 53 45 20 20 
Timed out.

command[S, 1]: 53 20 20 
<hangs here>
```

FYI, when plugging the device, dmesg shows 

```
[ 2430.663748] usb 1-2: new full-speed USB device number 5 using xhci_hcd
[ 2430.934711] usb 1-2: New USB device found, idVendor=2341, idProduct=0058, bcdDevice= 1.00
[ 2430.934713] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2430.934715] usb 1-2: Product: Arduino Nano Every
[ 2430.934716] usb 1-2: Manufacturer: Arduino LLC
[ 2430.934717] usb 1-2: SerialNumber: E611C9CE5151363448202020FF161E41
[ 2430.936620] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
```

Is the Arduino Nano Every supported? How do I start avarice in this particular case?

Florin-Popescu, 2021-02-12:
Hi,

I have only added support to the newer AVR families (so the UPDI protocol) via EDBG programmer (which uses the `jtag3` class). But it looks to me like JTAG2UPDI is actually based on JTAG MK2, like you said. This programmer uses the class `jtag2` in avarice, which would need to be modified to support the UPDI protocol.

So the answer would be that no, sadly it's not supported.

I can't really add the support for it since I don't have a device to test it on. But if you'd like to try you should be able to take the changes I've done for the `jtag3` class and port them to `jtag2`. They are somewhat detailed on the project's README and hopefully the implementation wouldn't be very different for the `jtag2`.

Florin-Popescu, 2021-06-13:
Closing this since it's getting fairly old. Please let me know if you need further assistance.

closed 2021-06-13 by Florin-Popescu