Florin-Popescu/avarice-unofficial #4: [ATTiny814] Cannot load ELF: wrong length for memory access

TimVosch, 2021-05-29:
Hi,

First of all thanks for your work on this! After a bit of struggling I could get a GDB server starting. (I did not have hidapi installed! `sudo apt install libhidapi-hidraw0 libhidapi-libusb0`) 

Unfortunately I am now running into a few issues with a device I am trying to add.

I tried to format and provide you with all the information you need. Let me know if you need something else 👍 

## What I am trying

I am trying to add the ATtiny814 and flash an ELF file

## How I added attiny814

- Used the io_reg.py script to create `attiny814_io_registers` in `ioreg.cc` and the header
- Added a device descriptor instance. (see below) 
  - `fuse_bitmap` is most likely incorrect
  - `OCD module` might be incorrect. `SYSCFG` is a 0x0F00, but I do not know the offset
- Flash / eeprom taken from datasheet (see screenshot). `avarice -k` reports the correct sizes

[image]

## What doesn't work

`load` in GDB causes avarice/atmel-ice to throw an error: `Wrong length for memory access`.

```
# terminal 1
$ ./src/avarice -4 -u -P attiny814 -d :4242
    <...see output below...>

# terminal 2
$ avr-gdb
> target ext :4242
> load firmware.elf
  Exit
```

Also notable that in gdb `info mem` shows a second region, in the avarice logs there is a memory-map XML that says its the RAM. But that would be incorrect since the ATtiny814 only has 512b SRAM.

[image]

## Code and logs

### Devdescr struct
```c++
// DEV_ATTINY814
    {
        "attiny814",
        0x9322, // From avrdude
        64,     // Page size FLASH
        128,    // No. of pages FLASH
        32,     // Page size EEPROM
        4,      // No of pages EEPROM
        26 * 2, // First flash address which is not an interrupt vector
        DEVFL_NONE,
        attiny814_io_registers,
        false,
        0x07, // Fuse bitmap (???)
        0,    // fuses
        0,    // osccal
        0,    // OCD revision
        {
            0 // no mkI support
        },
        {
            0 // no mkII JTAG support
        },
        {
            0 // no Xmega support
        },
        {
            fill_b2(0x8000), // Start address of Program memory
            128,             // Page size of flash in bytes
            32,              // Page size of EEPROM
            fill_b2(0x1000), // Address of NVMCTRL module
            fill_b2(0x0F80), // Address of OCD module (???)
        },
    },
```

### Avarice logs

```
Connection opened by host 10.0.0.7, port 50120.
GDB: <qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+>
->GDB: qXfer:memory-map:read+
GDB: <vMustReplyEmpty>
->GDB:
GDB: <!>
->GDB: OK
GDB: <Hg0>
->GDB:
GDB: <qTStatus>
->GDB:
GDB: <?>
->GDB: S05
GDB: <qfThreadInfo>
->GDB:
GDB: <qL1160000000000000000>
->GDB:
GDB: <Hc-1>
->GDB:
GDB: <qC>
->GDB:
GDB: <qAttached>
->GDB:
GDB: <g>

GDB: (Registers)Read 32 bytes from 0x800000
jtagRead
command "read memory" [0x12, 0x21]
0E 00 0E 00 12 21 00 20 00 00 00 00 20 00 00 00
Received 0x81 0x11 0x00 0x27 0x0e 0x0e
read:  0e 0e 00 12 84 00 00 00 8e 00 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Got message seqno 14 (command_sequence == 14)
response: 12 84 00 00 00 8E 00 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
jtagRead
command "read memory" [0x12, 0x21]
0E 00 0F 00 12 21 00 20 5D 00 00 00 03 00 00 00
Received 0x81 0x11 0x00 0x0a 0x0e 0x0f
read:  0e 0f 00 12 84 00 00 00 00 00

Got message seqno 15 (command_sequence == 15)
response: 12 84 00 00 00 00 00

command "read PC" [0x12, 0x35]
0E 00 10 00 12 35 00
Received 0x81 0x11 0x00 0x0a 0x0e 0x10
read:  0e 10 00 12 83 00 00 00 00 00

Got message seqno 16 (command_sequence == 16)
response: 12 83 00 00 00 00 00
PC = 0
->GDB: 00008e000000030000000000000000000000000000000000000000000000000000000000000000
GDB: <qL1160000000000000000>
->GDB:
GDB: <qXfer:memory-map:read::0,18a>
->GDB: l<memory-map>
  <memory type="ram" start="0x800000" length="0x20000" />
  <memory type="flash" start="0" length="0x2000">
     <property name="blocksize">0x40</property>
  </memory>
</memory-map>

GDB: <vFlashErase:00000000,00000080>
erasing 128 bytes @ 0x0

command "enter progmode" [0x12, 0x15]
0E 00 11 00 12 15 00
Received 0x81 0x11 0x00 0x06 0x0e 0x11
read:  0e 11 00 12 80 00

Got message seqno 17 (command_sequence == 17)
response: 12 80 00

command "chip erase" [0x12, 0x20]
0E 00 12 00 12 20 00 00 00 00 00 00
Received 0x81 0x11 0x00 0x06 0x0e 0x12
read:  0e 12 00 12 80 00

Got message seqno 18 (command_sequence == 18)
response: 12 80 00
->GDB: OK
GDB: <vFlashWrite:0:\x19\xc0 \xc0\x1f\xc0\x1e\xc0\x1d\xc0\x1c\xc0\x1b\xc0\x1a\xc0\x19\xc0\x18\xc0\x17\xc0\x16\xc0\x15\xc0\x14\xc0\x13\xc0\x12\xc0\x11\xc0\x10\xc0\x0f\xc0\x0e\xc0\x0d\xc0\x0c\xc0\x0b\xc0\x0a\xc0>
buffering data, 48 bytes @ 0x0
->GDB: OK
GDB: <vFlashWrite:30:\x09\xc0\x08\xc0\x11$\x1f\xbe\xcf\xef\xcd\xbf\xdf\xe3\xde\xbf\x02\xd0\x0b\xc0\xdd\xcf\xcf\x93\xdf\x93\x1f\x92\xcd\xb7\xde\xb7\x19\x82\x89\x81\x8f_\x89\x83\xfc\xcf\xf8\x94\xff\xcf>
buffering data, 46 bytes @ 0x30
->GDB: OK
GDB: <vFlashDone>
committing to flash
jtagWrite
command "write memory" [0x12, 0x23]
0E 00 13 00 12 23 00 C0 00 00 00 00 40 00 00 00 00 19 C0 20 C0 1F C0 1E C0 1D C0 1C C0 1B C0 1A C0 19 C0 18 C0 17 C0 16 C0 15 C0 14 C0 13 C0 12 C0 11 C0 10 C0 0F C0 0E C0 0D C0 0C C0 0B C0 0A C0 09 C0 08 C0 11 24 1F BE CF EF CD BF DF E3 DE BF 55
Received 0x81 0x11 0x00 0x07 0x0e 0x13
read:  0e 13 00 12 a0 00 35

Got message seqno 19 (command_sequence == 19)
response: 12 A0 00 35
Failed to write target memory space: wrong length for memory access

command "stop debugging" [0x12, 0x14]
0E 00 14 00 12 14 00
Received 0x81 0x11 0x00 0x06 0x0e 0x14
read:  0e 14 00 12 80 00

Got message seqno 20 (command_sequence == 20)
response: 12 80 00

command "AVR sign-off" [0x12, 0x11]
0E 00 15 00 12 11 00
Received 0x81 0x11 0x00 0x06 0x0e 0x15
read:  0e 15 00 12 80 00

Got message seqno 21 (command_sequence == 21)
response: 12 80 00

command "sign-off" [0x01, 0x11]
0E 00 16 00 01 11 00
Received 0x81 0x11 0x00 0x06 0x0e 0x16
read:  0e 16 00 01 80 00

Got message seqno 22 (command_sequence == 22)
response: 01 80 00
EXCEPTION: wrong length for memory access
```

Florin-Popescu, 2021-06-13:
Hey!

First of all regarding the fuse bitmap, it is a mask for which fuse "registers" are available on your device. So looking at ATtiny814's datasheet, it's 0x5F7 (0 on positions 0x3 and 0x9 since these are Reserved, and 1 on positions 0x0-0x2, 0x4-0x8, 0xA). Presumably it would throw an error if attempting to access a fuse which is unavailable based on this mask.
[image]

You also probably forgot to change the flash page size in the updi_device_desc part of the device descriptor. Correct value is 64 bytes, not 128. That is used by avarice when initializing the connection to the target device instead of the flash page size which is at the begining of the outer device descriptor struct. **May be why you are getting that failure.**

And regarding the OCD module address, I really can't remember why I set it to 0x0F80 for the ATmega4808. But it allows me to set breakpoints so I thought it was fine. Cannot find much info on neither that one nor on ATtiny814, so if it allows you to set breakpoints it should probably be ok too.

The RAM size reported to GDB using `info mem` appears to be hard-coded in avarice and is probably done so to account for all possible AVR RAM sizes. Also there is a comment when avarice outputs that xml saying:
`/* The RAM size indicated includes the possible EEPROM range, so GDB will treat EEPROM uploads  just like simple SRAM load operations. AVaRICE will disambiguate them based on their virtual offset. */`
So that size being incorrect probably doesn't hurt at all. GDB may try to access invalid memory due to this but avarice will probably reject such requests anyway.

Added your device to the necessary files in latest commit. Can you please pull that and retry? Also if there are further issues please post the full output of avarice since your log is missing the initialization messages between avarice and the target (log starts at command_sequence == 14).

TimVosch, 2021-06-13:
Hey Florin

Thanks for the help and the explanations. I pulled the changes and recompiled avarice. I am indeed able to load firmware on the device and set breakpoints. 🎉 

There's a few other issues, like avarice exiting suddenly and variables values not changing / initializing. Once i can consistently reproduce it, I will post more details. For now I am quite busy with graduating, so I will not much time in the upcoming week.

closed 2021-10-05 by Florin-Popescu

Florin-Popescu, 2022-05-20:
Hey, it's been almost a year since this was brought up. Just wanted to mention that I've discovered a bug giving me the same issue, which should be fixed now in v2.14-7. Give it a try if you are still interested.