Florin-Popescu/avarice-unofficial #7: New name or bump major

majioa, 2021-10-12:
Hello,

i saw you have forken avarice from official repo, and maintain this new repo.
As the packager of altlinux i have some troubles in current versioning, since you've downgrade major digit with keeping name of the project. To solve the trouble I propose 2 ways:
1. Bump major digit to 3, to differ from the main repo.
2. rename the project to for example "novel-avarice/newavarice/avaricee"

Florin-Popescu, 2021-10-15:
Would like to keep the versioning similar to upstream, to be obvious which upstream version this fork tracks. So I wouldn't like to bump major version. Is something small like bumping the suffix for the tags enough for you? I am thinking of something like `2.14-2`, `2.14-3` etc... because currently in ubuntu the package available via apt is `2.14-1`.

I found that the current official version in altlinux seems to be `2.13-alt1`. Is this correct? In this case, will a tag like described above be seen correctly as a bump?

If this won't work for you, what is the exact version string of the current package in your packaging system, and what is the syntax you use for comparisons?

majioa, 2021-10-16:
It is ok to keep the upstream versioning, but how about to the current versioning in the fork? Upstream last version is 2.14, but your one is 1.x, it is strange for me, if you plan to use in your versions, patch (third) number. i.e. 2.14.1, 2.14.2, 2.14.2.1 (for minor patch) etc. it will be ok.

> I found that the current official version in altlinux seems to be 2.13-alt1

Yes it is correct. If you do 2.14.1, 2.14.1.1 etx versioning, I can do bumps.

Florin-Popescu, 2021-10-16:
Renamed the tags like discussed.

majioa, 2021-11-22:
@Florin-Popescu Sorry for disturn again, have you a principle to numerate in format x.y-z , and not x.y.z ?

Florin-Popescu, 2021-11-22:
Well I like to follow the [debian versioning scheme](https://www.debian.org/doc/debian-policy/ch-controlfields.html#version) since I use Ubuntu. In that way it matches the format of the version which comes from my distro's packages and it looks like the generally accepted way as far as I can tell. I would like to stick to this.