Florin-Popescu/avarice-unofficial #8: Finish support of tinyAVR 1-series devices? (ATtiny1616 in particular, but I'd like to see them all supported)

urjaman, 2022-11-26:
In short, i have an ATtiny1616 (and Atmel-ICE) i could test with,
if someone would do the patch/PR for it.

Given how sparse the supported devices list for the 1-series is, i assume they've been
individually tested? (sparse as in 412 and 814)
Also, is there a particular set of tests to do on a new device?

If we can script the part support process sufficiently much, couldnt we
just add the whole series (and maybe the 0-series too) and just keep seperate record of what is tested?
Maybe do the flashrom chip support thing of requesting for works/doesnt reports from users, when using an untested chip.

Florin-Popescu, 2022-11-26:
Hello!

I'm sure such a script would be possible to develop but I don't have time for it. Maybe I'll find some time during the holidays but unlikely.

Until now I've only manually added the devices I need and merged PRs from others who did the same. To do it manually it's a pretty long list, since we now have the 0 and 1 series for both mega and tiny AVR, so 4 "families" total, although they are functionally identical from what I can tell. There are also some other devices missing from the upstream avarice, like the atmega324pb which I've used in the past.

If you are ok for now with just adding support for the attiny 1616 it's pretty easy to do and you can submit a PR yourself, or I can do that and you can help test it.

If you or anyone else would have a go at adding support for all the new AVR devices, I'd be of course happy to merge that. Then we could potentially ask to merge it upstream to Jorg's sources.

Florin-Popescu, 2023-08-15:
It's been almost a year and based on #13 I think you no longer need this. Closing.

closed 2023-08-15 by Florin-Popescu